XtheOne/Inverter-Data-Logger #1: Multiple inverters support in all plugins

XtheOne, 2017-09-06:
Current state is that it is only implemented for domoticz.
Other plugins will follow.

XtheOne, 2017-09-12:
PVOutput is updated.
Now you can define apikey and sysid for every inverter.
So multiple accounts or one donation account can be used for multiple inverters.
ScanLoggers can be used to detect the Logger(s) and Inverter(s) on your local network.

XtheOne, 2017-09-26:
Should work now

kajhoej, 2018-04-13:
I have 2 inverters connected and use the mqtt plugin but I cannot separate the 2 inverters in the mqtt output as they have the same subject. Shouldnt the inverters serial number be part of the subject in the mqtt message?

XtheOne, 2018-04-16:
Well, the mqqt plugin was written by someone else.
I can change the script in several ways.

I can change "power/solar/e_total" to "solar/<ID>/e_total" so inserting the inverter SN into the message path.
Or make separate entries in the config file like it is done for Domoticz and them make the publish path prefix configurable.
So what would you prefer?

XtheOne, 2018-04-16:
I can change this in 2 ways.
Change "power/solar/..." to "solar/<ID>/..."
Or make separate entries in the config file like the domoticz plugin. You could then select separate mqqt servers & paths.

XtheOne, 2018-04-25:
Multiple inverters support added in mqtt

closed 2018-04-25 by XtheOne