XtheOne/Inverter-Data-Logger #3: Add Version 5 protocol support

XtheOne, 2017-10-01:
Implement support for V5 protocol as used in new embedded ethernet loggers.
Needs documentation from Omnik and or iGEN / Solarman.

This is requested.
Also a firmware update file is requested.

XtheOne, 2017-10-12:
Development done in branch: "Experimental_Frame_Version_5_support"

abc1000, 2017-10-13:
Tried 00-ff for the first bit, some works for 70%..
But with these empty, i'v got also a error..
frame_hdr = ''
command = ''

XtheOne, 2017-10-13:
frame_hdr = byte(headCode) + short(dataFieldLength) + short(contrlCode) + short(serialNumber)
Change only contrlCode and serialNumber (which is a frame sequential number)
if you change command length then change dataFieldLength to match the nr of bytes.

XtheOne, 2017-10-30:
As there is still no more help from Omnik or SolarMan about the V5 frame data needed to trigger the logger we are a bit stuck at the moment.
Will now wait for updates for the Android Apps or firmware from/for the ethernet module as none of the apps now gets data from the new logger in local mode.
It seems that the firmware still is buggy as it frequently locks up so hopefully an update will come.

Phyxion, 2017-11-18:
I have a SolarMAN logger in 501xxxxxx S/N range. It supports the Android app just fine (in local mode) and sending and receiving _some_ data works with the script, but then this happens (with LiveStats.py). I've also tried the v5 branch, but it's the same issue, so I'm not sure if that's used. Would this even work? 

> 2017-11-18 12:47:27,716 DEBUG DATA len=85:
2017-11-18 12:47:27,722 INFO Inverter ID: S6XXXXXXXX
Traceback (most recent call last):
  File "LiveStats.py", line 17, in <module>
    inverter_exporter.run()
  File "/home/pi/InverterV5/Inverter-Data-Logger/InverterExport.py", line 150, in run
    self.logger.info("Inverter main firmware version: {0}".format(msg.main_fwver))
  File "/home/pi/InverterV5/Inverter-Data-Logger/InverterMsg.py", line 154, in main_fwver
    if (self.__get_int(101) == 0): return ""
  File "/home/pi/InverterV5/Inverter-Data-Logger/InverterMsg.py", line 33, in __get_int
    return int(self.raw_msg[begin:begin+1].encode('hex'), 16)
ValueError: invalid literal for int() with base 16: ''

FYI, this is sent:
```
2017-11-18 12:49:47,872 DEBUG RAW sent Packet (len=16): 68:02:41:b1:86:88:e5:1d:86:88:e5:1d:01:00:15:16  hA
2017-11-18 12:49:48,721 DEBUG RAW received Packet (len=99): 68:55:41:b0:86:88:e5:1d:86:88:e5:1d:81:01:0b:53:36:46:31:33:43:54:34:34:37:20:20:20:20:20:20:00:ab:06:f1:00:00:00:00:00:04:00:00:00:00:00:03:00:00:00:00:08:d1:00:00:00:00:13:8f:00:2d:00:00:00:00:00:00:00:00:00:25:00:00:df:55:00:00:39:02:01:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:12:16  hUAS6XXXXXXXX      -%U9
```

Phyxion, 2017-11-18:
Added a PR, thanks for this script 👍

XtheOne, 2017-11-20:
The reply is only 99 bytes, have to check the docs what frame type this is.

Phyxion, 2017-12-12:
FYI, the main_fwver and slave_fwver both still fail in InverterMsg with this logger. Most likely because 101 and 121 don't exist due to 99 bytes.

XtheOne, 2017-12-13:
Yes, cannot get data past EOF. Still need to fix this.
No time too many work to do before end of this year :-(

XtheOne, 2017-12-13:
Still NO help from Omnik or Solarman dev...

coraminformatica, 2018-02-09:
Hello, I need also the V5 support. 
How can I help you?
I tried your frame reader for V5 but I get right only the serial number.
I can't decode e_total, e_today or other important values.

Regards

Raamkonijn, 2018-02-15:
Here a S/N 502****** users with two inverters on one logger.
After installing the master-branch, disabling the printing of main_fwver and slave_fwver it does return some values.

InverterLib.py:
```
    @property
    def main_fwver(self):
        return ""

    @property
    def slave_fwver(self):
        return "" 
```

```
2018-02-15 16:13:41,828 DEBUG Importing output plugin ConsoleOutput
2018-02-15 16:13:41,829 INFO Connecting to logger with IP: 10.0.0.235 and SN 502******
2018-02-15 16:13:41,830 INFO connecting to 10.0.0.235 port 8899
2018-02-15 16:13:41,833 DEBUG RAW sent Packet (len=16): 68:02:41:b1:93:e9:f1:1d:93:e9:f1:1d:01:00:09:16  hA
2018-02-15 16:13:42,365 DEBUG RAW received Packet (len=99): 68:55:51:b0:93:e9:f1:1d:93:e9:f1:1d:81:03:04:30:30:30:30:30:31:31:32:38:31:34:33:30:30:30:31:01:0c:14:fb:00:00:00:00:00:05:00:00:00:00:00:05:00:00:00:00:10:12:00:00:00:00:01:f4:00:26:00:00:00:00:00:00:00:00:00:11:00:06:c8:a4:00:00:2b:47:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:6f:16  hUQ0000011281430001&+Go
2018-02-15 16:13:42,365 DEBUG DATA len=85:
2018-02-15 16:13:42,366 INFO Inverter ID: 00000***81430001
2018-02-15 16:13:42,366 INFO Inverter main firmware version:
2018-02-15 16:13:42,366 INFO Inverter slave firmware version:
2018-02-15 16:13:42,366 INFO RUN State: 4096
2018-02-15 16:13:42,367 DEBUG Run pluginConsoleOutput
Inverter ID: 00000***81430001
E Today :  17.0  Total: 44458.0
H Total : 11079   Temp : 26.8
errorMsg:     0
PV1   V: 537.1   I:  0.5
PV2   V:   0.0   I:  0.0
PV3   V:   0.0   I:  0.0
L1    P:    38   V: 411.4   I:  0.5   F:   5.0
L2    P:     0   V:   0.0   I:  0.0   F:   0.0
L3    P:     0   V:   0.0   I:  0.0   F:   0.0
```

I did change the divider of e_today in InverterLib.py

```
        def e_today(self):
        """Energy generated by inverter today in kWh"""
        return self.__get_short(69, 1)  # Divide by 1 instead of 100
```

I do miss the decimals from e_today, on the portal from solarmanpv.com it says:

```
PV1    543,4V    5,5A   L1   412,1V   5,5A    4.100W
PV2    556,8V    2,5A   L2   412,1V   5,4A    0W
PV3    0,0       0,0    L3   413,2V   5,4     0W

Frequency 50Hz
Today's Energy(kWh) 17,3
Total Energy(kWh) 44.458
Temperature 26,6
```

I modified MQTTOutput to parse the data as JSON. The 'power'-value says: 4.5 (4500W)

Now I'm stuck how to get the decimals right on the e_today, and possibly the voltage value's.

XtheOne, 2018-03-16:
Still........ Nothing from Omnik. :-(

stenjo, 2018-04-11:
I'm really interested in this. Have a data logger stick from Ginlong where they obviously are using frame format V5 as the inverter is logging new sets of data like total energy yesterday, this month and last month. Think I have debugged where to find the main firmware version and yesterdays data, but my energy reading does not match what my app claims to be the current production level.
Any way I can contribute?

XtheOne, 2018-04-16:
I would really need some trace data.
I still did not find what magic string would trigger the logger.
In the mean time I am working on a listen server which waits for a logger message and then processes the message. You can then add a server in the logger pointing to the python server IP/port

stenjo, 2018-05-02:
[Solis Three phase communication protocal.pdf](https://github.com/XtheOne/Inverter-Data-Logger/files/1967100/Solis.Three.phase.communication.protocal.pdf)
Got hold of this document. Will that cover Version 5 of the protocol?
Looks to be Version 7...

XtheOne, 2018-10-16:
I might get an ethernet logger myself later this year.
No replies from Omnik or iGEN until now.

nielsvn92, 2019-01-10:
Not sure if you're still working on this, but I tried your experimental branch in attempt to get my inverter working with your script, and I actually do get a response now. I didn't get that on the master branch.

This is the error I'm getting:
`2019-01-10 17:42:13,020 INFO Loggers found  on the network: ['192.168.1.xxx', '640xxxxxx'].
2019-01-10 17:42:13,020 INFO Connecting to logger with IP: 192.168.1.xxx and SN 640xxxxxx
2019-01-10 17:42:13,021 INFO connecting to 192.168.1.xxx port 8899
2019-01-10 17:42:19,127 DEBUG RAW sent Packet (len=15): a5:02:00:10:45:00:00:7b:8a:26:26:01:00:a9:15  E{&&
2019-01-10 17:42:19,164 DEBUG RAW received Packet (len=38): a5:19:00:10:15:00:18:7b:8a:26:26:01:01:29:da:0f:00:e1:98:00:00:c1:9b:27:5c:2b:45:52:52:3d:2d:31:0d:0a:0d:0a:f0:15  {&&)'\+ERR=-1
2019-01-10 17:42:19,164 DEBUG DATA len=640.0: 
Traceback (most recent call last):
  File "LiveStats.py", line 17, in <module>
    inverter_exporter.run()
  File "/<path>/Inverter-Data-Logger/InverterExport.py", line 137, in run
    if (msg.msg)[:9] == 'DATA SEND':
  File "/<path>/Inverter-Data-Logger/InverterMsg.py", line 88, in msg
    return self.__get_string(self.Frame_offset + 0, self.len + self.Frame_offset + 0)
  File "/<path>/Inverter-Data-Logger/InverterMsg.py", line 28, in __get_string
    return self.raw_msg[begin:end]
TypeError: slice indices must be integers or None or have an __index__ method
`

I have an Omnik 3K TL3, and the Wi-Fi data logger has firmware version H4.01.51MW.2.01W1.0.64(2018-01-251-D).
I too am not able to configure any other remote server.

ppiwowar, 2019-01-11:
After experimenting with our omnik/wifi we get the same error result. I think we have a problem with calling packet frame.

XtheOne, 2019-01-14:
The problem is that I do not have a new WiFi module or able to get a firmware update from Omnik.
I asked them several times for a firmware file.
This started with the new Ethernet module which had the first buggy implementation of the V5 protocol from iGEN. This interface crashes on the magic packet so we stopped there.
It looks that this new WiFi firmware also has V5 but is responding to the magic frame.
I will look what data it returns.
It is also strange that entering more than the default server is not possible anymore.
This is why the server script was made to also support in decoding the V5 frame layout.
Until now I only got some information of the frame layout from the android app. but the implementation there is incomplete and won't work.

XtheOne, 2019-01-14:
ERR=-1 probably shows that the V5 magic frame is not correct.
What does scanloggers returns?

ppiwowar, 2019-01-14:
We did not use scanloggers.py.  LiveStats.py with setting in the config file:
gateways = auto
did not discover any logger automatically. Only setting logger manually in the config file:
gateways = 192.168.1.160,638xxxxxx
found the logger and returned the frame. We will check with scan loggers.py too.
Could you please specify, what is the 'android app' you use so we could check it here. Standard Omnik app in the manual mode does not give the access to electricity parameters of inverter. It gives a possibility to change some config parameters ....
General question: Has Omnik may limited access to the inverter by third party software so the portal is the only option?

XtheOne, 2019-01-14:
iGEN is the company which creates the logger. they upgraded the V4 protocol to V5 last year.
The app on android I use is from Omnik and the one from iGEN which is newer and has some parts about V5 but incomplete.
Until now no luck on getting a firmware update file for the newer loggers...
Or any documentation.
I hoped that someone could enter my IP in their logger so I would get some frames to analyse.

nielsvn92, 2019-01-14:
What we are able to do is setting the internal server of port 8899 to `client` mode, and insert any IP/port as server. I've tried it before, but it seems there is no data coming out. If I remember correctly there was only the following TCP flow:
- Handshake
- Directly FIN, FIN-ACK (I believe from both sides, will have to double check)

So no data.

If you want, I could move my logger to another subnet and grant you access to the inverter WiFi module via a VPN @XtheOne? Or would physical access be required?
When I had the gateway set to auto, it could detect my logger btw with the correct SN.

Is there anything else we can try for you?

nielsvn92, 2019-01-14:
I used Wireshark to capture the data, and some program to use as a TCP server. It could of course be because of the server, because it doesn't send a message though. Not sure how the protocol works.

I have checked for firmware before, but could not find anything. But looking at XtheOne's messages, he has found firmware before. Perhaps a downgrade would work indeed.

XtheOne, 2019-01-15:
This implementation has a few tasks.
Decode V5 frames, these frames are the same as sent to the public site.
Connect to the local logger and trigger it to sent data by sending a magic frame.
Update listen server.

XtheOne, 2019-01-15:
Recently I asked for an update file but they asked me for the SN and a picture of my logger.
I did not yet respond as I don't have the newest logger.

nielsvn92, 2019-01-15:
If you have some way for me to send the magic frame to my local logger, e.g. run some Python script or something, I can show you the results from Wireshark, or even send you a capture file?

I can also supply you with the SN and picture of my logger if you'd like, in a DM.

XtheOne, 2019-01-15:
For testing I set my logger @home to sent data to my office and test the decoder remotely.
Under Advanced > Server B: 46.182.220.24 Port 10004/TCP
I understand now that this is not possible with these new loggers?
This would enable me to determine the offsets of the data block.

I don't know if the newer WiFi module is still based on the Hi-Fly HF-A11 module.

nielsvn92, 2019-01-15:
We have two different options, I might be able to share screenshots when I get home to clarify.
The options are (both under Advanced):
- Remote server
-- this is set to some default value which we can't change, and is probably the Omnik portal. We can't add a remote server as well.
- Port settings (by heart, have to double check the exact text)
-- this can be set to Client or Server. By default this is Server, and port 8899. We can also set this to client, and try and point it to your server (46.182.220.24:10004?)

Which one do you mean?

ppiwowar, 2019-01-15:
I am very much interested in the result of second option as I have tried it some time ago.
Setting: client with ipaddress and port. Receiving server (RaspberryPi) was indicating the connection but no further data coming in.... Maybe the protocol needs some starting frame to be sent?
Please tell me your result.

nielsvn92, 2019-01-15:
I just enabled the second option and selected your server and port. It should send data now, or at least try to.

ppiwowar, 2019-01-16:
Nielsvn92, have you got any data transmission?

nielsvn92, 2019-01-16:
I don't know. I can't sniff when I'm not at home. I also came to the conclusion that I'm not going to see the network traffic to another host than mine without doing something with monitor mode on my WiFi card.

I think it's best to check with @XtheOne if he has received data today from me on the specified host/port.

XtheOne, 2019-01-17:
Sorry guys, mu untangle server was down due to bad memory in that server.
It is up now and I will start the omnik server project now.

nielsvn92, 2019-01-18:
Please let me know if you at least receive some network traffic. Preferably data of course, but it's also good to know if you receive any TCP packets at all.

ppiwowar, 2019-01-18:
Nielsvn92, you may also want to direct the traffic to:
piwowar.dynalias.com:8989
My server is up & running all the times.

nielsvn92, 2019-01-18:
I can't send it to two servers, only one.

ppiwowar, 2019-01-18:
> No to XtheOne, since he's trying to figure out the data structure for us :p

OK. Understood. Is he getting any data ?Regards, Pawel
    W piątek,

nielsvn92, 2019-01-18:
I don't know, the only communication we have is via this ticket.

nielsvn92, 2019-01-18:
Perhaps it's easier to use Discord, instead of chatting in this issue?
I created a server which you can join: https://discord.gg/URK9pP

XtheOne, 2019-01-18:
I am working on this now and I just got to the point that I get the ERR=-1 after sending the magic frame.
Maybe a wireshark trace between the logger and the omnik portal can show hot the portal starts the data frame reply.
It looks that the magic packet is not ok.
I first got connected, but no data was sent.

loekd, 2019-02-07:
By adding an additional 0x00 to the endcode, making the frame 16 bytes in length I've managed to get a series of 29 byte messages back from the inverter. No idea yet about how to parse it. 
InverterLib.py ln 65, change into:
```
    endCode = '\x15\x00'
```
Additional perk; sending a 16 byte frame, the module no longer crashes.

XtheOne, 2019-02-08:
Interessting, it is not in the java code. But it is still not working.
Here a sniplet of the code which encodes a frame.

    public byte[] encode() {
        ChannelBuffer buf = ChannelBuffers.buffer(ByteOrder.LITTLE_ENDIAN, getLength());
        buf.writeByte(this.headCode);
        this.dataFieldLength = createDataFieldLength();
        buf.writeShort(this.dataFieldLength);
        buf.writeShort(this.contrlCode);
        buf.writeShort(this.serialNumber);
        buf.writeInt(this.sn);
        this.dataFields = createDataField();
        buf.writeBytes(this.dataFields);
        buf.writeByte(EncodeUtil.getCheckDigit(buf, 1, buf.readableBytes()));
        buf.writeByte(this.endCode);
        return buf.array();
    }

loekd, 2019-02-10:
@ppiwowar here you go:
```
"A5-10-00-10-15-00-05-0D-2E-E3-25-01-01-10-24-0C-00-25-6B-00-00-00-00-00-00-01-00-50-15"
"A5-10-00-10-15-00-06-0D-2E-E3-25-01-01-1E-24-0C-00-56-6B-00-00-00-00-00-00-01-00-90-15"
"A5-10-00-10-15-00-07-0D-2E-E3-25-01-01-20-24-0C-00-5D-6B-00-00-00-00-00-00-01-00-9A-15"
"A5-10-00-10-15-00-08-0D-2E-E3-25-01-01-21-24-0C-00-62-6B-00-00-00-00-00-00-01-00-A1-15"
```
@XtheOne based on that I can't say how many bytes are written into `buf`. I just figured it may crash because of a mismatch between expected frame length and actual length. This seems to be the case. Do you have code that parses the returned bytes?

ppiwowar, 2019-02-10:
What I see here is 4 x the magic packets, I guess, but not a response frame. Are you getting any response? You mentioned a frame with 29 bytes...?

Pozdrawiam/Regards, Pawel

loekd, 2019-02-10:
These are some 29 byte responses.

Sent from Nine<http://www.9folders.com/>

XtheOne, 2019-02-11:
This is what I make of these frames. Data seems only something with time, could be a TM frame.
headCode          A5 (-91)
dataFieldLength 10 00 (16)
contrlCode         10 15 (21/16)
serialNumber      00 05 (5)
sn                       0D 2E E3 25 (635645453)
dataField             01 01 10 24 0C 00 25 6B 00 00 00 00 00 00 01 00 (...$..%k........)
 (fr6)                   01-01-1E-24-0C-00-56-6B-00-00-00-00-00-00-01-00
 (fr7)                   01-01-20-24-0C-00-5D-6B-00-00-00-00-00-00-01-00
 (fr8)                   01-01-21-24-0C-00-62-6B-00-00-00-00-00-00-01-00
checksum           50 (80)
endCode            15 (21)

XtheOne, 2019-02-11:
contrlCode 0x1510 (1010100010000)
version = V5
frametype = 5
sendOrReplyCommand = 1 (Reply)

nielsvn92, 2019-04-04:
@XtheOne could you perhaps check Discord? You're missed there :-).

To update everyone on what we've tried to do as well:
I tried the Solarman app to connect to the inverter locally, but even the app can't connect. I have the same result as with the script in this repo.

pbouwen3, 2019-04-23:
@nielsvn92 any chance you could share the discord invite again?
It would be great to be able to follow up.

Thanks!

nielsvn92, 2019-04-24:
@pbouwen3 https://discord.gg/krjReA

peterzandbergen, 2019-05-13:
I have been sniffing the packets that the wifi module sends to the Omnik site. Will that information help? I see a lot of different sized packets being sent. Let me know.

lennardk, 2019-05-21:
Same as peter I have access to packet dumps, in my case for firmware H4.01.51MW.2.01W1.0.64(2018-01-251-D) connected to a Omnik3000tl.
Happy to join you guys on discord if that's still going on.

XtheOne, 2019-05-21:
I really have no time now to work on this project.
Maybe in a few weeks I would get some time but for now I have to get a big project done...

stingone, 2019-06-19:
Any update on this the V5 protocol? :)

MrClever, 2019-10-16:
While I'm waiting for this to be resolved, I've hacked an egregious obscenity that polls the inverter's web interface and scrapes the current power and energy production. This is then fed into a python script which cleans up the output a little more and does the upload to PVOutput...or you can roll your own to go somewhere else. The SoFar web interface (on my inverter) doesn't expose all the data available (*like volatages, current, grid AC freq. etc*) but it's better than nothing ¯\\\_(ツ)\_/¯

I take no responsibility for this; it was born out of desperation!

To see what else is exposed try `curl -s -o - --user admin:admin http://<your_inverter>/status.html`

If anyone is interested, I can share my Python PVOutput uploader too.
```
#!/usr/bin/env bash

username=admin
password=admin
inverter=127.0.0.1 # Change to your inverter's IP/hostname
curlOpts="-s -o - --user ${username}:${password}"

poll(){
	curl ${curlOpts} http://${inverter}/status.html 2>&1 |\
	egrep '^var webdata_(now|today)' |\
    	sed 's/var webdata_//' |\
		sed 's/["\;\=]//g' |\
		sed 's/\r$//g'
}

output=$(poll)

if [[ ! -z "${output}" ]]; then
	echo -e "${output} OK"
else
	echo "FAIL"
fi
```

droneando, 2019-12-23:
Hi! My wifi stick serial number starts with 0522xxxxx and I'm having the time outs responses from your script.
Any update in getting it to work with V5 protocol?
In the meanwhile do you think it's possible to extract the inverter data from the solarman web page where this info is shown to the user?

netadair, 2020-02-06:
I have tested it with a RHI inverter and found this so far:

- My serial number does not work when sending the request; any other willl do, very strange, so for time being I just add 1; answer will come anyway with the proper sn 
- Change the serial_hex code to cope with large serial numbers (like mine):
```
logger_sn_mod = logger_sn + 1
serial_hex = bytearray(hex(logger_sn_mod).rstrip("L").lstrip("0x"))
```

- Parsed request with comments
```
                # a5:           start 10100101
                # 02:00:        le data length (data)
                # 10:           ?
                # 45:           code?
                # 00:           ? (mirrored to answer)
                # 00:           counter
                # ea:d4:4c:ef:  le serial no
                # 01:00:        command? or target adress 0x1, unspecific=0x0 ?
                # 51:           chk
                # 15            eof 10101
```

- Parsed answer with comments
```
                # a5:           start
                # 10:00:        data length
                # 10:           ?
                # 15:           ?code?  answer to 45
                # 00:           ? (mirrored from request)
                # d2:           packet counter
                # ea:d4:4c:ef:  serial no in hex reversed
                # 01:01:        command? or address of inverter device? (modbus?)
                # e4:f0:52:00:  le, seconds;  operating time!?
                # 7e:71:00:00:  le, seconds; feed-in time today?
                # 73:49:e9:5d:  le,seconds; inverter setup starttime
                # 06:00:        ?
                # 1f:           chk
                # 15            eof
```

stravinci, 2020-03-05:
@XtheOne I got documentation for V5 from IGEN. Are you interested to implement it?

[69 more replies]