XtheOne/Inverter-Data-Logger #14: Usage of CSV file with jsunnyreports

ervee, 2018-01-16:
Hi!

I'm successfully using your application to pull data from my Omnik and store the CSV output in a unique file every 1 minute. But I'm unable to load the CSV files into jsunnyreports (http://www.jsunnyreports.com/). Problem is, I don't know what values jsunnyreports is expecting at what position in the file... Perhaps you have experience with this?

I also contacted the creator of jsunnyreports and awaiting response. Perhaps more people have issues with this so this issue would perhaps also help other people once we figure this out and post results here.

Thank you.
Ralf.

Jendemen, 2018-03-15:
Hi ervee,

How did you manage to get the CSV output?
Store data to MySQL is working, but I prefer the CSV output.

Thanks in advance, Jan

XtheOne, 2018-03-19:
Enable the CSVOutput plugin.

Jendemen, 2018-03-19:
Of course I enabled the CSVOutout plugin, but no data.
Using the CSVOutput2file is working now! Thanks.

For my other inverter (SMA) I use SBFspot, which creates a daily csv file and (hopefully) als a month totals csv file. CSVOutput2file is only using 1 file and constantly adding new data to it?

XtheOne, 2018-03-21:
Yes, only one file. No logic yet to create daily files.
Since no data is stored no monthly stats are possible.
Use Domoticz export or SQL and extract your own tables.

XtheOne, 2018-03-21:
Added option to create a daily CSV file.
Update your config file.

Jendemen, 2018-03-21:
GREAT!  I will have a look.  Thanks again.

XtheOne, 2018-03-22:
One last note, CSVOutout plugin outputs the data to the console, not a file...

closed 2018-03-22 by XtheOne