XtheOne/Inverter-Data-Logger #15: Problems with loggin

hlubach, 2018-03-03:
After install the files in the Subdir  /storage/Omnik, and making the config file.
Gave the command  python LiveStats.py and this is the result I get.

LibreELEC:~/Omnik # python LiveStats.py
2018-03-03 11:48:16,238 DEBUG Importing output plugin ConsoleOutput
2018-03-03 11:48:16,241 INFO Connecting to logger with IP: 192.168.178.115 and SN 602538088
2018-03-03 11:48:16,242 INFO connecting to 192.168.178.115 port 8899
2018-03-03 11:48:16,252 DEBUG RAW sent Packet (len=16): 68:02:41:b1:68:00:ea:23:68:00:ea:23:01:00:df:16  hAh#h#
2018-03-03 11:48:16,902 DEBUG RAW received Packet (len=99): 68:55:41:b0:68:00:ea:23:68:00:ea:23:81:02:01:4e:4c:44:4e:32:30:32:30:31:33:32:53:32:30:36:37:00:f5:0b:32:00:00:ff:ff:00:0f:00:b6:ff:ff:00:0f:ff:ff:ff:ff:08:e4:ff:ff:ff:ff:13:90:01:69:ff:ff:ff:ff:ff:ff:ff:ff:00:58:00:01:4a:ed:00:00:4d:4d:00:01:00:00:00:00:ff:ff:00:00:00:00:00:00:00:00:00:00:70:16  hUAh#h#NLDN2020132S20672iXJMMp
2018-03-03 11:48:16,903 DEBUG DATA len=85: 
2018-03-03 11:48:16,904 INFO Inverter ID: NLDN2020132S2067
Traceback (most recent call last):
  File "LiveStats.py", line 17, in <module>
    inverter_exporter.run()
  File "/storage/Omnik/InverterExport.py", line 150, in run
    self.logger.info("Inverter main firmware version: {0}".format(msg.main_fwver))
  File "/storage/Omnik/InverterMsg.py", line 154, in main_fwver
    if (self.__get_int(101) == 0): return ""
  File "/storage/Omnik/InverterMsg.py", line 33, in __get_int
    return int(self.raw_msg[begin:begin+1].encode('hex'), 16)
ValueError: invalid literal for int() with base 16: ''
LibreELEC:~/Omnik # 

What do I wrong

XtheOne, 2018-03-16:
Less bytes are returned than I got in the specs so it fails on getting the firmware data.
Need to fix this...

XtheOne, 2018-03-19:
Fixed
Can you check now?

hlubach, 2018-03-19:
So far as i can see  it works .
In my config i have PVoutputOutput  en the log fil, but after python
LiveStats.py i see te data is send , but the log file stays clean

Han

XtheOne, 2018-03-21:
LiveStats.py is only to check if you get data from the inverter, it forces to only use consoleoutput plugin.
You run InverterExport.py to really sent the data to the chosen plugins.

hlubach, 2018-03-21:
han@han-Satellite-L875 ~/Downloads/Inverter-Data-Logger-master $ python
InverterExport.py
2018-03-21 17:13:28,705 DEBUG Importing output plugin PVpoutputOutput
Traceback (most recent call last):
  File "InverterExport.py", line 202, in <module>
    inverter_exporter.run()
  File "InverterExport.py", line 66, in run
    __import__(plugin_name)
ImportError: No module named PVpoutputOutput

XtheOne, 2018-03-22:
The correct plugin name is: PVoutputOutput

hlubach, 2018-03-22:
Overlooked  that ,
changed it , but now this one

han@han-Satellite-L875 ~/Downloads/Inverter-Data-Logger-master $ python
InverterExport.py
2018-03-22 11:06:46,570 DEBUG Importing output plugin PVoutputOutput
Traceback (most recent call last):
  File "InverterExport.py", line 202, in <module>
    inverter_exporter.run()
  File "InverterExport.py", line 66, in run
    __import__(plugin_name)
  File "/home/han/Downloads/Inverter-Data-Logger-master/outputs/PVoutputOutput.py",
line 3, in <module>
    from pytz import timezone
ImportError: No module named pytz
han

XtheOne, 2018-03-22:
pip install pytz

hlubach, 2018-03-22:
All working now.
Had no pip on the laptop easy install , but googled how to install pip
And Voila

with thanks
Han

XtheOne, 2018-03-22:
Great that it works now.

closed 2018-03-22 by XtheOne