XtheOne/Inverter-Data-Logger #17: Error in configuration file

donhay, 2018-03-24:
When I run python LiveStats.py I get the following reported. So it can see my inverter and it reports the values.

pi@raspberrypi:~/Inverter-Data-Logger $ python LiveStats.py
2018-03-24 16:50:18,228 DEBUG Importing output plugin ConsoleOutput
2018-03-24 16:50:18,230 INFO Connecting to logger with IP: 192.168.0.120 and SN 616039440
2018-03-24 16:50:18,231 INFO connecting to 192.168.0.120 port 8899
2018-03-24 16:50:18,822 DEBUG DATA len=89: 
2018-03-24 16:50:18,823 INFO Inverter ID:   S2821590450130
2018-03-24 16:50:18,824 INFO RUN State: 1
2018-03-24 16:50:18,824 DEBUG Run pluginConsoleOutput
Inverter ID:   S2821590450130
E Today :   1.8   Total: 2177.7
H Total :  5774   Temp : 18.0 
errorMsg:     0
PV1   V: 233.4   I:  0.1
PV2   V:   0.7   I:  1.9
PV3   V:   0.0   I:  0.0
L1    P:    44   V: 238.2   I:  0.3   F: 50.05
L2    P:     0   V:   0.0   I:  0.0   F:   0.0
L3    P:     0   V:   0.0   I:  0.0   F:   0.0
2018-03-24 16:50:18,827 DEBUG DATA len=17: 
2018-03-24 16:50:18,828 DEBUG Exit Status: DATA SEND IS OK

If I then manually run python InverterExport.py I get the following reported.
It tells me there is an error in the config file with no section for my inverter ID.

pi@raspberrypi:~/Inverter-Data-Logger $ python InverterExport.py 
2018-03-24 16:51:24,503 DEBUG Importing output plugin DomoticzOutput
2018-03-24 16:51:24,553 INFO Connecting to logger with IP: 192.168.0.120 and SN 616039440
2018-03-24 16:51:24,554 INFO connecting to 192.168.0.120 port 8899
2018-03-24 16:51:27,330 DEBUG DATA len=89: 
2018-03-24 16:51:27,332 INFO Inverter ID:   S2821590450130
2018-03-24 16:51:27,333 INFO RUN State: 1
2018-03-24 16:51:27,333 DEBUG Run pluginDomoticzOutput
2018-03-24 16:51:27,334 ERROR no section in configuration file for inverter with ID:   S2821590450130, skipping.
2018-03-24 16:51:27,335 DEBUG DATA len=17: 
2018-03-24 16:51:27,336 DEBUG Exit Status: DATA SEND IS OK

Here is my config file.

###############
### Settings ###
################

[general]
# General:enabled_plugins
# Choose which outputs to use
# Possible options: MysqlOutput,PostgreSQLOutput,PVoutputOutput,ConsoleOutput,CSVOutput,CSVOutput2File,DomoticzOutput,MQTTOutput
enabled_plugins = DomoticzOutput

# choose one of these
# use temperature of inverter for output to pvoutput/mqtt
use_temperature = false
# do not use temperature of inverter for output to pvoutput/mqtt
#use_temperature = false

# Minimum Temperature for exporting. (a max. of 300 degrees is hardcoded)
# Set to -32768 to always sent the temperature value.
# Or set to a value so that the inverter internal temperature is only exported if equal or above this value.
min_temp = 5

# Minimum Voltage of any sensor for exporting.
# Set to 0 to always sent the sensor's voltage value.
# Or set to a value so that the sensor voltage data is only exported if equal or above this value.
min_voltage = 0

# Minimum grid Frequency for exporting.
# Set to 0 to always sent the sensor's frequency value.
# Or set to a value so that the grid Frequency is only exported if equal or above this value.
min_freq = 30

[logger]
# ip,sn,ip,sn,ip,sn or auto
# IP address of your inverter(s) data logger and S/N of the wifi kit(s)
gateways = 192.168.0.120,616039440
# Automatically detect logger(s) on the local network
#gateways = auto
# Default for the Wifi module/logger/gateway
port = 8899
# time to wait for inverter logger response
timeout = 3

### repeat this for every inverter ID that must be logged to pvoutput
### The apikey(s) and sysid(s) can be found at http://pvoutput.org/account.jsp
[pvout]
#use apikey-<inverter serial> in the apikey name
#apikey-NLBN1234567A1234 = NOTAREALAPIKEY86e2258d4e29169fb79cf18b00
#use sysid-<inverter serial> in the sysid name
#sysid-NLBN1234567A1234  = 12345

### use domoticz-<INVERTER SERIAL> as section name
### repeat this section for every inverter ID that must be logged to domoticz
### Replace 'idx' with the Idx number of the corresponding virtual sensor device as defined in domoticz
### comment out values which can or must be skipped
[domoticz-S2821590450130]
host = 192.168.0.100
port = 8080
path = /json.htm
# Replace 'idx' with the Idx number of the corresponding virtual sensor device as defined in domoticz
Electric_meter_idx         =4
# comment out below values which are not needed
#Temp_idx                   = idx
#E_today_idx                = idx
#E_total_idx                = idx
#E_total_c_idx              = idx
#H_total_idx                = idx
### DC input
PV1_voltage_idx            =5
PV1_current_idx            =6
#use below when you have multiple PVs on your inverter
#PV2_voltage_idx            = idx
#PV2_current_idx            = idx
#PV3_voltage_idx            = idx
#PV3_current_idx            = idx
#PV123_voltage_idx          = idx
#PV123_current_idx          = idx
### AC output
AC1_voltage_idx            =7
AC1_current_idx            =8
AC1_power_idx              =9
AC1_frequency_idx          =10
#use below when you have a 3-phase inverter
#AC2_voltage_idx            = idx
#AC2_current_idx            = idx
#AC2_power_idx              = idx
#AC2_frequency_idx          = idx
#AC3_voltage_idx            = idx
#AC3_current_idx            = idx
[mysql]
# Host where the mysql server is active
host = 127.0.0.1
user =
pass =
database =

[postgresql]
connstr = host=10.0.0.1 dbname=inverter user=inverter password=example

[mqtt]
# Host where the mqtt server is, including user/pass authentication
host = 127.0.0.1
port = 1883
user = username
pass = password

[csv]
daily_file = false
disable_header = false
csv_file_name = filename.csv

[log]
# Log:Output
# Possible options: none,console,file (combinations are possible)
# Use none to disable logging
type = console

# Log:level
# Possible options: critical, error, warning, info, debug, notset
level = debug

# Log:filename
# Output file for file logger
filename = inverter-export.log

I don't see what is wrong so could someone please point me in the right direction.
I feel I'm so close, but just don't understand the scripts.

XtheOne, 2018-03-30:
I have to fix the data hex dump in debug mode...
Can you check if the inverter ID is correct? it might be missing the last digits...

donhay, 2018-03-31:
I have checked the label on the inverter itself and the ID is S2821590450130. This is the same as in my config file.

Do you need any more information from me?

Thank you for your help.

donhay, 2018-04-04:
Sorry for being a noob, not sure what you mean by the raw data dump. If you need more than the following please let me know.

After running InverterExport.py I now get the following.

pi@raspberrypi:~/Inverter-Data-Logger $ python InverterExport.py
2018-04-04 09:00:17,653 DEBUG Importing output plugin DomoticzOutput
2018-04-04 09:00:17,702 INFO Connecting to logger with IP: 192.168.0.120 and SN 616039440
2018-04-04 09:00:17,703 INFO connecting to 192.168.0.120 port 8899
2018-04-04 09:00:17,710 DEBUG RAW sent Packet (len=16): 68:02:41:b1:10:04:b8:24:10:04:b8:24:01:00:d5:16  hA$$
2018-04-04 09:00:18,316 DEBUG RAW received Packet (len=103): 68:59:51:b0:10:04:b8:24:10:04:b8:24:81:02:0a:20:20:53:32:38:32:31:35:39:30:34:35:30:31:33:30:00:aa:09:7c:00:07:00:00:00:06:00:19:00:00:00:07:00:00:00:00:09:85:00:00:00:00:13:86:00:8f:00:00:00:00:00:00:00:00:00:14:00:00:56:b4:00:00:17:08:00:01:00:00:00:00:07:97:0a:5b:00:00:00:01:00:00:00:00:00:00:00:00:46:16  hYQ$$  S2821590450130|V[F
2018-04-04 09:00:18,317 DEBUG DATA len=89: 
2018-04-04 09:00:18,318 INFO Inverter ID:   S2821590450130
2018-04-04 09:00:18,319 INFO RUN State: 1
2018-04-04 09:00:18,320 DEBUG Run pluginDomoticzOutput
2018-04-04 09:00:18,321 ERROR no section in configuration file for inverter with ID:   S2821590450130, skipping.
2018-04-04 09:00:18,322 DEBUG RAW received Packet (len=31): 68:11:51:f0:10:04:b8:24:10:04:b8:24:44:41:54:41:20:53:45:4e:44:20:49:53:20:4f:4b:0d:0a:23:16  hQ$$DATA SEND IS OK#
2018-04-04 09:00:18,323 DEBUG DATA len=17: 
2018-04-04 09:00:18,324 DEBUG Exit Status: DATA SEND IS OK

XtheOne, 2018-04-04:
This is what I need. (DEBUG RAW received Packet (len=103): <raw data>)
Will try to fix it by running it trough the test script.
What type of inverter do you have?

XtheOne, 2018-04-04:
I see, there are 2 spaces before the S/N starts. I see: "  S2821590450130" as the SN.
Just commited a fix.

XtheOne, 2018-04-05:
This seems to be good now.

closed 2018-04-05 by XtheOne