XtheOne/Inverter-Data-Logger #20: Using LAN data logger

alekslyse, 2018-07-17:
This is a bit strange problem - I have a set of WIFI stick and LAN stick (looks the same) and wanted to use the LAN sticks for security reasons as it seem like its not possible to turn off the AP on the WIFI sticks.

After some restarts I managed to get one MQTT output from the inverter, but most of the times it fails with this - seem like its possible to get it working, just something that doesnt make sense:

2018-07-17 18:09:07,790 DEBUG Importing output plugin ConsoleOutput
2018-07-17 18:09:07,790 INFO Start listening on IP: 0.0.0.0 and Port 10004
2018-07-17 18:09:07,790 INFO Use CTRL-Break to exit.
2018-07-17 18:09:07,791 DEBUG waiting for a connection.
2018-07-17 18:09:08,257 DEBUG connection from: ('192.168.1.151', 64470)
2018-07-17 18:09:57,793 DEBUG RAW received Packet (len=14): a5:01:00:10:47:6e:08                                                              :95:84:69:71:00:c1:15  Gniq
2018-07-17 18:09:57,794 DEBUG DATA len=1:
2018-07-17 18:09:57,795 INFO Inverter ID:
Traceback (most recent call last):
  File "InverterServer.py", line 199, in <module>
    inverter_exporter.run()
  File "InverterServer.py", line 150, in run
    self.logger.info("RUN State: {0}".format(msg.run_state))
  File "/home/aleks/Inverter-Data-Logger/InverterMsg.py", line 138, in run_state
    return int(self.__get_short(offset[14], 1))
  File "/home/aleks/Inverter-Data-Logger/InverterMsg.py", line 78, in __get_shor                                                              t
    num = struct.unpack('!H', self.raw_msg[begin:begin + 2])[0]
struct.error: unpack requires a buffer of 2 bytes

XtheOne, 2018-07-18:
The simple reason is that the LAN interface uses protocol V5 which is not yet supported.
I have little info from Omnik & Solarman about this new protocol.
So it is still under development. I will get a LAN interface later this year.
Then I will be able to debug & trace the new packet format.

alekslyse, 2018-07-18:
Anything I can do for you to help you make it work?

alekslyse, 2018-07-20:
> Yes, If you can let your logger sent it's data to my test system I could work out the packet layout.

Please let me know the host and ip

XtheOne, 2018-10-16:
Got no reply so closing this one.

closed 2018-10-16 by XtheOne