XtheOne/Inverter-Data-Logger #22: New inverter + wifi module stops the script from working

doppiej, 2018-10-25:
I had a faulty inverter and received a new one including a new wifi module. The wifi module number starts with 0637, I am not seeing this one in your supported list.
I updated the script with the new inverter and when I run the script it actually finds the wifi module and then everytime the script runs it times out and doesn't find any data.
Do you know what I can do about it?

XtheOne, 2018-11-07:
Hmm, it seems that it does not respond to the magic frame.
Can you check with the app in local mode if it gets any data?
It could that it uses the new V5 protocol but then you have to trace the data stream to the logging server.
Does the server script work?

Arnaud48, 2018-11-18:
Same issuess her, sn 0637000009 , with my first omnik sn 1600.... no problem and now  stuck.

XtheOne, 2018-11-19:
Since I don't want to buy a new inverter :-) you can help me by letting your inverter sent log data to my IP so that I could look to the data frames.
I think that the logger has new firmware which uses a newer data frame protocol which I partially decoded from the android app.
I will ask iGEN / OMNIK again for newer firmware.
Please mail me for my IP and instructions.

ppiwowar, 2019-01-03:
Same issue here (S/N = 638xxxxxx). Moreover, there is only one Remote Server available(A), fixed for the portal. No choice for B or C. Have you been able to find a proper magic frame?

pbouwen3, 2019-04-23:
> Same issue here (S/N = 638xxxxxx). Moreover, there is only one Remote Server available(A), fixed for the portal. No choice for B or C. Have you been able to find a proper magic frame?

Same here!
Would like to help out by sending data to @XtheOne, but no choice to add new remote server...