XtheOne/Inverter-Data-Logger #28: Any idea how to handle an Omnik with SN 640xxxxxx ?

hermanvanderhoek, 2019-02-11:


XtheOne, 2019-02-11:
The protocol has changed. When the script was developed the Frame format was V4.
iGEN has moved to Frame format V5.
iGEN and Omnik did not help me in any way up till now.
I have some clues from the iGEN android app but V5 is not working in this app also. Only partial implementation.

hermanvanderhoek, 2019-02-11:
Hmmm, that's a pity.... Was on my way to write my own app in Delphi :-( Thanks for the reply any way!

XtheOne, 2019-02-11:
I am still trying but the first step is to figure out the format of the frame which triggers the logger to return current inverter data. Parsing the data itself is not that difficult.

hermanvanderhoek, 2019-02-11:
I found the thread about V5 support.....Reading it now...

stingone, 2019-06-19:
> I found the thread about V5 support.....Reading it now...

Where did you found the thread about V5 support... i also got SN 640xxx