XtheOne/Inverter-Data-Logger #31: GoodWe 55000DNS the target machine actively refused it

JamesBacon, 2019-06-15:
2019-06-15 11:24:03,779 ERROR Could not open socket
2019-06-15 11:24:03,779 ERROR [Errno 10061] No connection could be made because the target machine actively refused it
2019-06-15 11:24:03,780 ERROR Error connecting to logger with IP: 192.168.1.2 and SN , trying next logger.

Uksa007, 2020-10-24:
I get similar errors with a Goodwe 5000DNS and 3000XS
2020-10-24 16:32:33,860 INFO connecting to 192.168.60.118 port 8899
2020-10-24 16:32:33,864 ERROR Could not open socket
2020-10-24 16:32:33,865 ERROR [Errno 111] Connection refused
2020-10-24 16:32:33,866 ERROR Error connecting to logger with IP: 192.168.60.118 and SN removed, trying next logger.

Any fix for these loggers?

JamesBacon, 2020-11-03:
Figuring out what Wi-Fi kit you have and the Serial Number of it might help you. Documentation says you might find the Wifi Kit S/N on one of the sides of the Inverter and it should be 8 Digits long and start with a 6. I can not find mine but you might have more luck. Then in the Config file add your ip and then your SN. "192.168.1.101,60012334"

ThinkPadNL, 2021-02-08:
I would also like to get this working on my 3000XS. Any progress @Uksa007 ?
What i don't understand is how i should interpret the S/N that is talked about in this repository.

My Goodwe 3000XS has a serial number like this `x3000SSxaaaWxxxx` (sanitized). Are the last 8 characters what i need to fill in?

By the way, i did an `nmap` scan for 8899 and 48899 (both UDP) but with no results.

ThinkPadNL, 2021-02-08:
Ok, i tried this Python script and it discovered my wifi-module. But it gives me `[Errno 111] Connection refused`.

Uksa007, 2021-02-09:
> I would also like to get this working on my 3000XS. Any progress @Uksa007 ?

I didn't have any luck getting the wifi module to reply on any of the know ports, most of them UDP.
Not sure if these modules have the ability to communicate directly.

I ended up using a script to pull the data from the SEMS portal.
https://github.com/markruys/gw2pvo

Janvier123, 2023-01-13:
Using this on a GW5048D-ES

**NONE of the following works:**

ip + full SN
ip + full number from the sticker from the wifi kit
ip + last 8 digits starting with a 6  number from the sticker from the wifi kit
ip + wifi ssid number