XtheOne/Inverter-Data-Logger #34: PVOutput plugin does not work with low frequency updates

silvester747, 2019-09-11:
I am using InverterServer.py as an endpoint for my Omnik inverter. The inverter pushes data once every 5 minutes. The PVOutput plugin has a condition that it only sends the data to when the current minute is divisible by 5. So if my inverter pushes the data any other minute, it will not be reported.

I propose to make this configurable. For inverters that report more frequently, or when using polling, the update can be postponed until the 5th minute, for other inverters the update can be sent immediately.

conradhagemans, 2019-11-24:
I removed the 5 min check in the script. The omnik WiFi module sends the data any way every 5 min.

conradhagemans, 2019-11-25:
I like your changes. Will change my script accordingly. Not sure if this repo is still maintained.