XtheOne/Inverter-Data-Logger #36: Issues Omnik inverter with SN: 80911****

FlorisHigh3IT, 2020-03-15:
I used to read the data with this script for showing in my Node Red Dashboard. Worked perfect!

After my Omnik inverter died, this week I got a new Omnik2500tl. Unfortunately now it does not work anymore. Any idea how to solve this? 

This is the result on LiveStats.py:

python LiveStats.py
2020-03-15 13:15:38,539 DEBUG Importing output plugin ConsoleOutput
2020-03-15 13:15:41,693 INFO Loggers found  on the network: ['192.168.0.76', '80                                                                                                             9110293'].
2020-03-15 13:15:41,693 INFO Connecting to logger with IP: 192.168.0.76 and SN 8                                                                                                             0911****
2020-03-15 13:15:41,694 INFO connecting to 192.168.0.76 port 8899
2020-03-15 13:15:41,696 DEBUG RAW sent Packet (len=16): 68:02:41:b1:15:0b:3a:30:                                                                                                             15:0b:3a:30:01:00:09:16  hA:0:0
2020-03-15 13:15:44,699 ERROR Timeout connecting to logger with IP: 192.168.0.76                                                                                                              and SN 80911****, trying next logger.

gpvelzen, 2020-04-22:
+ 1 the same issue only my sn starts with 6427*****

FlorisHigh3IT, 2020-04-23:
I started using this plugin in Domoticz with succes: https://github.com/sincze/Domoticz-Omnik-Local-Web-Plugin

sincze, 2020-04-23:
Nice ;-)

gpvelzen, 2020-04-23:
For now I use this script, it  uploads it directly to pvoutput.org
https://github.com/Woutrrr/Omnik-Data-Logger/issues/27#issuecomment-490990351