XtheOne/Inverter-Data-Logger #37: Solarman logger LSW-3

Superi, 2020-04-30:
Is this logger type supported?

I have Sofar Solar inverter with this logger.

Auto search find my logger ip and serial, send packet but after few seconds i got timeout and no response. Tried to set timeout ~10sec but no help.

Logger Serial is 170xxxxxxx

fufs, 2020-06-26:
+1

I just had a Sofar Solar inverter installed and it came with this module. I don't feel like registering for the solarman app (although my data is being sent to their servers anyway...) and would love to have this data show in Home Assistant. I did a full port scan of the module and it seems that two ports are open - 80 and 8899. However as @Superi mentioned it timeouts. I also tried to set up the InverterServer and connect to it using the remote server settings tab, but it doesn't want to connect. Serial number starts with 172

If you need any data required to implement this module into this software, I'm happy to help.

fufs, 2020-06-28:
#3 
It seems that this logger uses V5 protocol which is not easy to use. Will have to wait for others to crack it.

glynnallen1704, 2020-07-11:
@Superi Same issue for me with Sofar Inverter mine also starts with 170******

awsumco, 2020-10-30:
@Superi Same issue on a sunsynk with serial of 173******* which use's solarman.

ronaldlangeveld, 2020-11-02:
Same issue, serial number 171*****

osman-masood, 2020-12-27:
My logger's serial number starts with 173 as well. Totally clueless on what the protocol is :(

michu79, 2021-05-04:
hallo my SN 1717
and log 
2021-05-04 19:50:32,695 DEBUG Importing output plugin DomoticzOutput
2021-05-04 19:50:32,721 INFO Connecting to logger with IP: 10.0.102.35 and SN 1717XXXXXX
2021-05-04 19:50:32,722 INFO connecting to 10.0.102.35 port 8899
2021-05-04 19:50:32,725 DEBUG RAW sent Packet (len=36): a5:17:00:10:45:00:00:74:c3:5c:66:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:01:03:00:00:00:27:05:d0:67:15  Et\f'g
2021-05-04 19:50:34,734 DEBUG RAW received Packet (len=29): a5:10:00:10:15:00:2a:74:c3:5c:66:02:01:eb:d4:e7:00:aa:08:00:00:7f:b4:a9:5f:05:00:f3:15  *t\_
2021-05-04 19:50:34,734 DEBUG received msg body (len=2): f3:15
2021-05-04 19:50:34,735 DEBUG DATA len=243:
2021-05-04 19:50:34,735 INFO Inverter ID: ?
2021-05-04 19:50:34,735 INFO Inverter main firmware version:
2021-05-04 19:50:34,736 INFO Inverter slave firmware version:
2021-05-04 19:50:34,736 DEBUG Run pluginDomoticzOutput
2021-05-04 19:50:34,736 ERROR no section in configuration file for inverter with ID: ?, skipping.

fufs, 2021-08-05:
For anyone interested I have a ghetto solution that I've been using for about a year now. `status.html` is a static page that is rendered every time you pull it. By doing so at an interval you get an HTML with all current values. Then you can parse the Vars from its script using sth like Node-RED. 

This solution uses the fact the status.html is properly whitespaced (even if it wasn't you could still get it to work but way harder)

vawg, 2021-08-05:
I am interested in how this works, sounds like an easter egg :)

fufs, 2021-08-05:
I once told it to a professional programmer and he laughed

cybermerlo, 2021-08-05:
Sadly in status.html there are only a few values (not everything that the logger record). Also, for whatever reason, all my datalogger's values are zeros, even if the multiples data values sent to SERVER A are, instead, correct.

MichaluxPL, 2021-09-19:
I've managed to get data from Sofar KTL-X series intverter via LSW-3.
Thanks to work by https://github.com/jlopez77.
If You are interested - let me know :)

asiviero76, 2022-07-25:
> I've managed to get data from Sofar KTL-X series intverter via LSW-3. Thanks to work by https://github.com/jlopez77. Working solution is here: https://github.com/MichaluxPL/Sofar_LSW3
> 
> Actually each inverter has it's own register address map. So @jlopez77 solution works great in terms of V5 protocol compatibility. It just requires different registers mapping and pointing registers position in the script code :)

Hi, anyone know how to get data from this inverter ( HYD 3000-ZSS HP ) that use LSW3 ?

charbelmalek, 2022-08-01:
i was able to connect to solarman datalogger for deye 5k inverter using https://github.com/MichaluxPL/Sofar_LSW3, i have response from the inverter but the data returned are wrong. any ideas ?

sigxcpu76, 2022-10-09:
Has anything been done for logger serial 23**** ?

baukgfdsde, 2022-11-03:
Yes, i have serial 23*** - you can turn on verbose=1 in config.cfg, then python3 InverterData.py tells you more about communication to wifi-logger LSW-3. In my case, there ist no register2 (only values below 0x0027) - i have turned off chunk 1 in the script.

sigxcpu76, 2022-11-03:
I ended up doing it from scratch in Go. Got my hands on the modbus registers list. It works flawlessly since then.

kubaceg, 2022-11-04:
@sigxcpu76 could You share code? And registers list?

sigxcpu76, 2022-11-04:
Registers list I've used is here: https://docs.google.com/spreadsheets/u/0/d/1p6p7n8-V2jlFid3FPPk5L4Sn4-tOWscQwpkahq5jRbs/htmlview#

The software (not ideal, was made by stealing bits & pieces from another one I did for a Infinisolar inverter) is attached.

[sofar-export.zip](https://github.com/XtheOne/Inverter-Data-Logger/files/9936164/sofar-export.zip)

kubaceg, 2022-11-04:
@sigxcpu76 Nice, at least something working with my inverter. Thanks

sigxcpu76, 2022-11-04:
Beware that it doesn't run on ARM (e.g. Raspberry) or any other different endianess than x86. I've made a bug in conversions or some of the libraries I am using does that, too lazy to dig. It failed on my RPi but it works on my x86 machine. All you need is to run `go build` (install golang if you don't have it) and use the output `sofar` binary with parameters.

kubaceg, 2022-11-04:
Ok maybe tomorrow I'll look at this RPI related problem. Do You have GH repo with this code? I want create fork. Or maybe You don't have nothing against me creating public repo with Your code?

sigxcpu76, 2022-11-04:
You can do whatever you want with it. I have my own private Git installation that I use. If you want, you can create a GH repo but you need to adjust the module path and the import paths. It is under "do whatever you want with it" license if you want.

kubaceg, 2022-11-12:
@sigxcpu76 I fixed it. Problem was with casting serial number to uint, for some reason it doesn't fit on ARM (atoi returns nice error in this occasion). So i rewrite config into yaml and during config initialization serial number is set as uint. Latest code version is on untagged `master` branch https://github.com/kubaceg/sofar_g3_lsw3_logger_reader. Please move further discussion there and feel free to contribute :)

tota-5, 2023-09-20:
just letting anyone know that sofar basic config works out of the box with refusol inverters!