XtheOne/Inverter-Data-Logger #44: LSW-3 SN 177xxxxxxx

ilexpl01, 2021-08-31:
Hi to all,

Has anyone got it working ?

I can discover the logger but not able to get any data from it:

root@NAS01:/volume1/@appstore/domoticz/var/scripts/Inverter-Data-Logger# python LiveStats.py
2021-08-31 09:36:01,738 DEBUG Importing output plugin ConsoleOutput
2021-08-31 09:36:04,750 INFO Loggers found  on the network: ['192.168.100.230', '1774324593'].
2021-08-31 09:36:04,750 INFO Connecting to logger with IP: 192.168.100.230 and SN 1774324593
2021-08-31 09:36:04,750 INFO connecting to 192.168.100.230 port 8899
2021-08-31 09:36:04,754 DEBUG RAW sent Packet (len=16): 68:02:41:b1:71:0b:c2:69:71:0b:c2:69:01:00:43:16  hAqiqiC
2021-08-31 09:36:14,762 ERROR Timeout connecting to logger with IP: 192.168.100.230 and SN 1774324593, trying next logger.

Can anyone help ? 

Need to import the data to my Domoticz HA.

BR
Mikolaj

MichaluxPL, 2021-09-19:
Yep :)
I've just managed to make it work for Sofar inverter.

MichaluxPL, 2021-09-20:
Working solution: https://github.com/MichaluxPL/Sofar_LSW3
Enjoy :)

closed 2024-08-01 by ilexpl01