XtheOne/Inverter-Data-Logger #47: LSW5 logger

roboneci, 2023-03-02:
Hello,

anyone can fetch data from logger LSW5 (sn: 350xxxxxxx) ?
Will some network dump help for debugging this logger?

Thanks for any answer.

roboneci, 2024-06-07:
Hello psunny28,

Could you please send me how you get those data? Thanks

psunny28, 2024-06-07:
# Remote Server Configuration

### Prerequisites
1. **Router with DNS Option**: Ensure your router supports DNS configuration.
2. **Domain Mapping**: Add your desired domain and map it to your PC's local IP address.

### Configuration Steps
1. **Access Logger Settings**:
   - Navigate to the logger's IP address (e.g., `http://10.10.0.80`).

2. **Server Configuration**:
   - You will see a page similar to the one below:
   
[image]

   
   - Configure `Server B` with the following details:
     - **IP Address**: Your system's local IP address (e.g., `10.10.0.100`).
     - **Domain Name**: The domain you set in your router's DNS settings (e.g., `iot.test.io`).
     - **Port**: `10000`.
     - **Connection**: `TCP`.

3. **Running the TCP Server**:
   - Start a TCP server on your local machine. Below is a sample TCP server code in Python:

    ```python
    import socket

    # Define the server address and port
    server_address = ('0.0.0.0', 10000)

    # Create a TCP/IP socket
    sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)

    # Bind the socket to the address and port
    sock.bind(server_address)

    # Listen for incoming connections
    sock.listen(1)
    print(f"Server listening on {server_address[0]}:{server_address[1]}")

    while True:
        # Wait for a connection
        connection, client_address = sock.accept()
        try:
            print(f"Connection from {client_address}")

            # Receive the data in small chunks and retransmit it
            while True:
                data = connection.recv(16)
                print(f"Received: {data}")
                if data:
                    connection.sendall(data)
                else:
                    break
        finally:
            # Clean up the connection
            connection.close()
    ```

### Notes
- After changing the settings, the device must be restarted.
- After the restart, you will need to re-login to the configuration interface.
- It is recommended to complete all configurations before restarting.

psunny28, 2024-06-07:
Got some more data too
@roboneci 

```
byte_sequences = [
    b'\xa5\x82\x00\x10A\x00\x03\x8cn\x91\xd1\x02\n\xca\x06\x00x\x00\x00\x00\x00\x00\x00\x00\x05<x\x016\x03LSW5BLE_17_DD01_1.23\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00@*\x8f?\xa2\x1c192.168.0.80\x00\x00\x00\x00\x03\x00\x01\x01\xdd\x07\x00\xffV1.1.00.0B\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xf7\x15',
    b'\xa5\x82\x00\x10A\x00\x02\x8cn\x91\xd1\x02\xeb\xc9\x06\x00Y\x00\x00\x00\x00\x00\x00\x00\x05<x\x016\x03LSW5BLE_17_DD01_1.23\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00@*\x8f?\xa2\x1c192.168.0.80\x00\x00\x00\x00\x03\x00\x01\x01\xdd\x07\x00\xffV1.1.00.0B\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xb7\x15',
    b'\xa5\x82\x00\x10A\x00\x01\x8cn\x91\xd1\x02\xcc\xc9\x06\x00:\x00\x00\x00\x00\x00\x00\x00\x05<x\x014\x03LSW5BLE_17_DD01_1.23\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00@*\x8f?\xa2\x1c192.168.0.80\x00\x00\x00\x00\x03\x00\x01\x01\xdd\x07\x00\xffV1.1.00.0B\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xffv\x15',
    b"\xa5\x04\x01\x10B\xd9\xa5\x8cn\x91\xd1\x01\x01\xdd\x1a\xd2\x06\x00\x88\x08\x00\x00\xc08\\f\x01\x00\x84\x03\x00\x001115C231211A0111PVBLINK         PVBS6K-M1       \x01\x03>\x00\t\x00\x16\x17p\x02\x00\x01\x00\x00'\x10\x00\x00\x17\x8e'\x10\x00\x00\x00|\x00\x00\x00\x00\x00\x07\nR\x02W\x01\xc6\x00\x00\x00\x00\x00\x00\x00\x00\x08\xd9\x00\x04\x04@\x06\xb1\x00\x05\x03\xaa\x00\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00$\x00\x00\x00\x00\x00\x00\x01z\x00\x00\x00\x00\x00\x00\x8cf\x00\x05j&\x07\xe8\x06\x07\x123\x18\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\t[\x13\x89\x00U\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05(\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05l\x00\x00\x05k\xff\xff\xff\xdb\xff\xff\xff\xce\xab\x15",
    b"\xa5\x04\x01\x10Bs\x95\x8cn\x91\xd1\x01\x01\xdd\xf6\xd1\x06\x00d\x08\x00\x00\xc08\\f\x01\x00\x84\x03\x00\x001115C231211A0111PVBLINK         PVBS6K-M1       \x01\x03>\x00\t\x00\x16\x17p\x02\x00\x01\x00\x00'\x10\x00\x00\x17\x8e'\x10\x00\x00\x00|\x00\x00\x00\x00\x00\x07\nR\x02W\x01\xc6\x00\x00\x00\x00\x00\x00\x00\x00\x08\xd9\x00\x04\x04@\x06\xb1\x00\x05\x03\xaa\x00\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00$\x00\x00\x00\x00\x00\x00\x01z\x00\x00\x00\x00\x00\x00\x8cf\x00\x05j&\x07\xe8\x06\x07\x123\x18\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\t[\x13\x89\x00U\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05(\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05l\x00\x00\x05k\xff\xff\xff\xdb\xff\xff\xff\xce\xec\x15",
    b"\xa5\x04\x01\x10Bs\x94\x8cn\x91\xd1\x01\x01\xdd\xd6\xd1\x06\x00D\x08\x00\x00\xc08\\f\x01\x00\x84\x03\x00\x001115C231211A0111PVBLINK         PVBS6K-M1       \x01\x03>\x00\t\x00\x16\x17p\x02\x00\x01\x00\x00'\x10\x00\x00\x17\x8e'\x10\x00\x00\x00|\x00\x00\x00\x00\x00\x07\nR\x02W\x01\xc6\x00\x00\x00\x00\x00\x00\x00\x00\x08\xd9\x00\x04\x04@\x06\xb1\x00\x05\x03\xaa\x00\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00$\x00\x00\x00\x00\x00\x00\x01z\x00\x00\x00\x00\x00\x00\x8cf\x00\x05j&\x07\xe8\x06\x07\x123\x18\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\t[\x13\x89\x00U\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05(\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05l\x00\x00\x05k\xff\xff\xff\xdb\xff\xff\xff\xce\xab\x15"
]
```

psunny28, 2024-06-07:
I was able to find some data for logger frames @roboneci 

{
    "Start": "a5",
    "Length": 130,
    "Control Code": "10 41",
    "Serial": "00 03",
    "Logger Serial": "35xxxxxx",  // Reversed bytes to show the original serial number format
    "Frame Type": "02",
    "Status": "0a",
    "Total Working Time": 338899928,
    "Power On Time": 0,
    "Offset Time": 0,
    "Modbus RTU Frame": {
        "Part 1": "05 3c 78 01",  // To be determined based on context
        "Part 2": "36 03",  // To be determined based on context
        "Serial Number": "LSW5BLE_17_DD01_1.23",  // Decoded string
        "Zeros": "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00",  // Padding zeros
        "Data": "40 2a 8f 3f a2 1c",  // Hex data that might need further interpretation
        "IP Address": "192.x.x.x",  // Decoded IP address
        "Zeros 2": "00 00 00 00",  // Padding zeros
        "Unknown Data 1": "03 00 01 01",  // To be determined based on context
        "Unknown Data 2": "dd 07 00 ff",  // To be determined based on context
        "Version": "V1.1.00.0B",  // Decoded version string
        "Trailing Zeros": "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"  // Padding zeros
    },
    "Checksum": "ff",
    "End Code": "f7 15"
}

sirhaplo, 2024-06-11:
Have you any idea of what register is send by the logger ?

psunny28, 2024-06-13:
No I am not able to find any register values @sirhaplo 

I was able to get and capture some packages which solarman logger was sending to server via proxy call.

sirhaplo, 2024-06-15:
What firmware do you have ?
I have `LSW5BLE_17_840B_1.05`
840B is labeled "Sensor list" in the app.
Is 33803 in big endian and 2948 as little endian, but i got no answer reading those registers.

psunny28, 2024-06-21:
I connected My inverter company and they said that register are stored in CRT sequence in inverter,
We can not directly get the values from the Inverter.

We need to find that way to get that data.

psunny28, 2024-07-03:
https://github.com/StephanJoubert/home_assistant_solarman/issues/614

I see you were able to reverse proxy and get the frame data,

were you able to successfully decode the frame ?

sirhaplo, 2024-07-04:
Finally i've read the registers from my , I did a massive brute force on all register and finally got some answers.

I used the registry_scan script here : https://github.com/jmccrohan/pysolarmanv5/blob/main/examples/register_scan.py

I scanned only holding registrer 500 by 500.
from 0 to 500, from 500 to 1000, and so on.
If i asked more registers it fails.

Finally i got 3 big groups of registers with some usefull data that i decode in this way :
- Open solarman smart, go to "Me" in the bottom toolbar and choose Local mode
- Connect via bluetooth to the logger using logger serial number or qr code
- Now you have realtime data
- Launch the scanner with the range that you found and search a very specific value
- values are very often scaled, so for SOC % of 58.1 you could search for 58 and found a registry of value `0581`

So the LSW5 stick is not a problem, the problem were my registers that are on very different ranges.

My inverter is a Renon Power IFL06, a clone of Megarevo R06LNA.
I started a new template for that inverter and i've got some results :
https://github.com/sirhaplo/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/renon_hybrid_iflXX.yaml

roboneci, 2024-07-04:
Hello,

i decompile solarman apk, and in directory assets there are files like that:
local_deye_5406_ble_params.txt
local_deye_5406_ble_realtime.txt
local_deye_5406_wifi_params.txt
local_deye_5406_wifi_realtime.txt
local_deye_5411_overview.json
local_deye_5411_params.json
local_deye_5411_realtime.json

and look what is in this file.
[local_deye_5411_realtime](https://github.com/roboneci/solarman/blob/main/local_deye_5411_realtime.json)

sirhaplo, 2024-07-04:
In my inverter/logger informations i found 2 interesting codes :
product_type : 1282
sensor_list : 840B

There are some file with those codes in the filename ? 
Could it possibly be the way to find the registers for every inverter

roboneci, 2024-07-04:
Download xapk from [here](https://apkpure.com/solarman-smart/com.igen.xiaomaizhidian/downloading)
open it with 7zip. In this xapk open com.igen.xiaomaizhidian.apk\assets\
and there are 3 files starting with 840b....json

I have sensor list 8420, but there is no such file :/
Module Version No：LSW5BLE_17_8420_1.2F
only one file matches local_invt_ble_activity_test.xml
but this file is DBase 3 data file which i can't open.

sirhaplo, 2024-07-04:
Thank you, i've done and I have found those files.
I will try to create a full adapter for my inverter.

In your case, can you reach the inverter with the local mode in bluetooth as i explained some message before ?
Because those configs seems to be how the app organize data in that interface

psunny28, 2024-07-04:
I tried to connect via bluetooth but it say `it does not support` Do I am again back to square 1.

psunny28, 2024-07-04:
@roboneci your and mine logger are directly sourced and they might not have Bluetooth connections enabled

[image]

But I see you have a case in code for Local Connection 

[image]

And it goes to Case 11.

I think you should try Bluetooth mode and sniff bluetooth connection transfer data via debugger logs on android.

Kmotr, 2024-09-07:
Does anybody have a LSW5 Firmware file? I'd like to have a look inside.

netadair, 2024-09-07:
@Kmotr if Sofar branded fw is ok, then I can provide you some LSW§ or LSW5 images...

Kmotr, 2024-09-08:
Sure, Sofar is ok.

netadair, 2024-09-08:
Hope it helps...
[LSW3_15_270A_1.72.bin.txt](https://github.com/user-attachments/files/16922478/LSW3_15_270A_1.72.bin.txt)
You can download old (Solarman, before migration to Solis cloud) Solis stick images at [MW3_15_0501_1.24.bin](http://47.254.36.66/0_D0002_18/MW3_15_0501_1.24.bin) , please also check https://github.com/dasrecht/deye-firmware/ .
I have some other images as well,  ME_0D, ME_0E, ME_08, these are LSE-3 and I think LSW-x, but I don't remember which is which actually.